aderusha/MQTTCarPresence #1: Will this work with D1 mini (non-pro)?

Snipercaine, 2018-02-20:
Hi,
Love this project.  I just tried it today and so far haven't been able to get it working.  I'm using a non-pro D1 mini.  I wouldn't expect that to be the problem. 
I modified the values at the top of the sketch with my info. The max packet size is 512. The sketch loads up fine without any errors. 
In Home Assistant I added the automation. Changed the entity_id to match the mqttNode (I assumed that was necessary, maybe it isn't). 
Most concerning is that when I plug the D1 into the computer, I don't see anything coming up in the Serial Monitor.  

Any ideas of something else I can try? 
Thanks!

aderusha, 2018-02-20:
This for sure will work on the non-pro mini (however the STL for the case may not work).  Serial for sure should be working @ 115200 8N1, so double check that the serial device is being created on your computer when plugging in the D1 mini and confirm you can get serial data to display.  From there you should be able to view debug messages regarding connection to WiFi and MQTT.  Next, use an MQTT client to make sure that the device is successfully publishing messages to your broker.  Finally, confirm that Home Assistant MQTT Discovery has been enabled in your `configuration.yaml` so that the `binary_sensor` device can be auto-created.

Snipercaine, 2018-02-21:
Thanks for the quick response! 
I got it working.  I didn't have the mqtt discovery loaded.  
I also did the automation in the automation editor.  

I'd like to make a how-to video for this project.  I'll give you all the credit. 
You can see the other stuff I've done here:  YouTube.com/c/DrZzs 

Good work on this!  Genius idea! 
Thanks
-Justin

ericflores, 2018-02-23:
As an alternative, I flashed Tasmota binary into the wemos using the flashing executable (Windows) that comes in the Espeasy zip file. Then configured following the instructions in the Tasmota wiki.
It is working perfectly. 

Yet, I would like to capture uptime in minutes or seconds, and Tasmota only provides hours.

When I leave home I need the gate to close quickly after disconnecting, but when arriving I need some time to step out of the car and exit the garage thru the gate. I know that if the Wemos have been up more than 5 minutes is because I'm arriving. I could setup Homeassistant to behave differently when I'm leaving based on uptime.

ericflores, 2018-02-23:
Ohh, I had not even read your code. Your are reporting uptime in milliseconds! Looks like I'll have to compile and install your code!

edif30, 2018-05-03:
Will a NodeMCU board work?

1043717432, 2018-05-04:
Thanks for sharing

gorinie, 2018-05-20:
Hi
It is working form me with an ESP-01 and an USB adapter for ESP-01. This is very useful, you just plug it in the usb port on your car. No additional antenna but a Wemos d1 Mini with [(https://github.com/martin-ger/esp_wifi_repeater)] to repeat the signal nearby the Garage Door.  Also my automation in Home Assistant make use of a threshold on the signal with a binary_sensor, in such a way one can trim the closing of the Garage with [https://github.com/marthoc/GarHAge]  even before disconnection (this requires tuning...)

yaml code for Home Assistant
```
binary_sensor:
 - platform: threshold
    entity_id: sensor.yarissignal
    lower: -85.0
    name: yaris-threshold  
    friendly_name: Yaris Threshold     
    hysteresis: 0.0
```

regards,
Edoardo

1043717432, 2018-05-25:
@gorinie  Hi, Hello, thanks for sharing. What kind of esp-01 do you use with USB interface?

gorinie, 2018-05-25:
Hi,
I bought on amazon these: 

https://www.amazon.it/AZDelivery-Adattatore-ESP-01-Modulo-Arduino/dp/B078J7LDLY/ref=sr_1_fkmr0_1?ie=UTF8&qid=1527231984&sr=8-1-fkmr0&keywords=usb%2Btal%2Besp01&th=1

Zoltant7, 2018-09-14:
This is  fun project, but i rather use HA with discovery feature so i only need my phone to detect that I'm home or not!

gorinie, 2018-09-14:
Did you try to convert your project in [esphomeyaml](https://esphomelib.com/esphomeyaml/) ? would be quite useful....

closed 2020-01-15 by aderusha