aderusha/MQTTCarPresence #4: Home Assistant not discovering D1 Mini Pro

FutureTense, 2018-08-13:
I’m using [this Aideepen WEMOS D1 Mini Pro - 16M Bytes External Antenna Connector ESP8266 WIFI IoT Board](https://www.amazon.com/dp/B07333L9VR/ref=cm_sw_r_cp_tai_WiqCBb5H7C2W1) but Home Assistant is not discovering it.  I know it’s connecting to my WiFi, because my router shows it and I can ping it.

What can I do to debug this?

aderusha, 2018-08-13:
Check to confirm that [Home Assistant MQTT Discovery](https://www.home-assistant.io/docs/mqtt/discovery/) is enabled, and try monitoring your MQTT broker messages using MQTT Lens (or similar) to make sure the device is sending messages.

neirbomn, 2019-01-16:
@aderusha,
I'm having considerable trouble getting this project to work. It's a fabulous idea and if I can get it to work it will be even better!

The binary_sensor 'CarPresence' and uptime and signal sensors are being created so MQTT discovery is definitely on.
I've also used MQTT Lens to check topics are being sent. They are as shown by a power cycle of the Wemos D1 Mini Pro in the screenshot below but I think my automation or HomeAssistant is where the problem lies for me.
[image]
In HomeAssistant, the binary sensor is shown all in lower case in the states section of the UI.
[image]

I altered the automations.yaml file entry for this but it doesn't seem to work. Any ideas appreciated - I've just about given up for the time being as I've spent 3 days trying to get it to work.

I have tried renaming the mqttNode in lowercase, this appears to either cause a conflict and the binary_sensor is repeatedly kicked off the MQTT server, setting the binary_sensor to OFF, it then reconnects which causes the binary_sensor to be set to ON and so on... I think this is if the 'CarPresence' binary_sensor is still in the config or if I send empty payloads and remove the 'CarPresence' sensors, update and reflash the WEMOS and also clear out MosQuiTTo's db and I check for the topic "homeassistant/binary_sensor/carpresence/state" then nothing is sent.

Is there anything you can think of checking or suggest?

aderusha, 2019-01-16:
Can you share your automation?

closed 2020-01-15 by aderusha