aderusha/MQTTCarPresence #5: New Binary sensor not being created

madzur, 2018-09-30:
Hello Aderusha and all. I love this idea and would really like to get this one working. I saw Dr. ZzS video and write up, which led me to this Github page.
I followed Aderusha's instructions with some help from Dr. ZzS more detailed steps and managed to set up Home assistant MQTT discovery and load Aderusha's sketch onto a D1 mini. The trouble starts when I plug the D1 mini into power and it connects to my wifi. I can see it in HA MQTT log as connected, however it is not adding itself as a binary sensor to configuration.yaml 

> 1538340853: New connection from 192.168.1.160 on port 1883.
1538340853: New client connected from 192.168.1.160 as CarPresence (c1, k15, u'').

Specifically, in Dr. ZzS writeup (here: http://drzzs.com/mqtt-car-presence/) I’m stuck at the following step: 

> After that, restart Home Assistant and plug in your D1 Mini. After a few seconds the blue LED will start flashing like crazy. This sketch is genius. It adds itself as an entity to Home Assistant.  Go to your states page and you should see a new “binary_sensor.CarPresence”.  Once you see that, you’re ready for the next step.

However the binary_sensor never shows up, either in configuration.yaml or in the states. Any tips on what I can do?

madzur, 2018-10-04:
I managed to get the serial monitor working and it is showing me the following output:
`
SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc

Hardware initialized, starting program load
Connecting to WiFi network: XXXXscandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt 

connected with XXXX, channel 11
dhcp client start...
ip:192.168.1.160,mask:255.255.255.0,gw:192.168.1.1
.
WiFi connected successfully and assigned IP: 192.168.1.160
Initialization complete

Attempting MQTT connection to broker: 192.168.1.101
MQTT discovery connectivity config: [homeassistant/binary_sensor/CarPresence/config] : [{"name": "CarPresence", "device_class": "connectivity", "state_topic": "homeassistant/binary_sensor/CarPresence/state"}]
MQTT discovery connectivity state: [homeassistant/binary_sensor/CarPresence/state] : [ON]
MQTT discovery signal config: [homeassistant/sensor/CarPresence-signal/config] : [{"name": "CarPresence-signal", "state_topic": "homeassistant/sensor/CarPresence-signal/state", "unit_of_measurement": "dBm", "value_template": "{{ value }}"}]
MQTT discovery signal state: [homeassistant/sensor/CarPresence-signal/state] : -56
MQTT discovery uptime config: [homeassistant/sensor/CarPresence-uptime/config] : [{"name": "CarPresence-uptime", "state_topic": "homeassistant/sensor/CarPresence-uptime/state", "unit_of_measurement": "msec", "value_template": "{{ value }}"}]
MQTT discovery uptime state: [homeassistant/sensor/CarPresence-uptime/state] : 20831
MQTT connected
pm open,type:2 0`

Still, when I check the states item in Home assistant, no new binary sensors appear. I’m out of my depth here, so any tips would be greatly appreciated.

aderusha, 2018-10-12:
Excellent, thanks for posting and tracking this down!

closed 2018-10-12 by aderusha