aderusha/MQTTCarPresence #6: Use this without the HASS automatic detection

Hyukishi, 2019-03-22:
Is there any chance you could rewrite a portion of this that would NOT utilize the automatic discovery of HASS? I want to be able to test the MQTT functionality without having to rely on HASS to pick up the device in order to be used and simply using the MQTT broker, in my case mosquitto, instead.

Villhellm, 2019-03-22:
You can subscribe to the topics manually as well. The topics are all listed in the ino file.

aderusha, 2019-03-23:
@Villhellm has it correct - the auto discovery is just there to help if you want to use it, there's nothing saying you must do so.  If you want to manually create an MQTT sensor, or use this with OpenHAB, or whatever else you're looking to do, you can do so without modifying any code in this project.  Just watch the MQTT messages, and configure your automations to react as needed.

closed 2019-03-23 by aderusha

Hyukishi, 2019-03-23:
Thanks to both of you. Have a good weekend!