aderusha/MQTTCarPresence #9: MQTT Retain

broyuken, 2020-01-15:
Hello,

I have this set up and working and LOVE it, it works great. The only issue I have is that when I restart HA the sensor state goes into unknown until I start the car again. How would I go about setting the retain flag or similar to make HA remember its last state upon a restart?

aderusha, 2020-01-15:
If you check out lines [152-157](https://github.com/aderusha/MQTTCarPresence/blob/master/MQTTCarPresence/MQTTCarPresence.ino#L152-L157) you'll find a group of commands to publish configuration and state to MQTT.  [The publish() function](https://pubsubclient.knolleary.net/api.html#publish2) accepts an optional third parameter for setting the retain flag.  Add a `, true` to the end of the function calls which don't already have that, and those additional published messages will also be retained.

broyuken, 2020-01-23:
Finally got a chance to try it, it worked perfect! Thank you

broyuken, 2020-01-23:
One thing I've noticed is that throughout the day when the car is off, the RSSI signal strength changes significantly. Any idea why this might be happening? The car is sitting in the garage off and it goes from -50 to -70. This throws off my card that shows if the car is in the garage or if it's away,

aderusha, 2020-01-23:
It's simply reporting the value provided to it by the ESP8266.  You're talking about a radio that was sold for a dollar, so adjust your expectations about the stability of that number accordingly.

If you wanted, one could possibly add a filtering library to smooth out the values over time with the hope of possibly removing noise from the signal.  This might be a good starting point: https://github.com/denyssene/SimpleKalmanFilter

broyuken, 2020-01-24:
That’s not what I’m talking about though. If I pull into my garage at 9am, and the RSSI is -50. I shut the car off and don’t get in it again that day. Around 11 or 12 the RSSI changes to -70. Note that the ESP has not turned on since 9am. This only started occurring since I enabled the retain flag on those 3 lines.

aderusha, 2020-01-24:
In that case something is publishing that value.  Try logging MQTT messages on your broker and see if you can't find what sent that value over.

broyuken, 2020-01-24:
Ok, I'll try to check my mosquitto logs

broyuken, 2020-02-18:
I checked mqtt explorer and it's retaining a value of -44 even though when my wife left for work this morning it had a value of -78. So when I restart HA it thinks my car is in the garage, when in fact it's gone. Something about the message isn't being retained correctly, any ideas? Either it's not updating to a new retained value or something.

aderusha, 2020-02-18:
Is there any chance one or more devices might have retain on, while another doesn't?  This should be trackable if you leave MQTT Explorer running and check the message history.

broyuken, 2020-02-18:
I'm not sure I follow, I have 2 cars I track with this, one has retain on (Odyssey), the other one I haven't updated yet (Murano). I wanted to fix this one first. Both devices have separate names, so they should be 2 separate entities.

aderusha, 2020-02-18:
Just throwing out ideas as to what might be happening there on your network, thought being maybe you have two devices publishing on the same topic but one of them isn't retaining for whatever reason.

I'd recommend keeping a copy of MQTT Explorer open (or similar), and using that to try and catch the problem in the act.  Those messages are there because something sent them, so it seems like the next step is to try and figure out what that was and then why.

broyuken, 2020-02-18:
So my wife just got home, the signal was set to -49, I restarted HA and now it's set to-64. However mqtt explorer is still showing -49. No idea where it's getting these numbers from.

kbweathe, 2020-03-03:
I just completed this project.  How can I make my garage door close as soon as I turn off the ignition. I have checked my automation in HA and I don't have a delay set.  Is the delay in the last will and testament?  If so how do I change it.  Help would be appreciated.

aderusha, 2020-03-03:
The LWT timeout is set by the "keepalive" value, which you can change in the pubsubclient.h file.  You can find details about this [here](https://github.com/knolleary/pubsubclient#limitations) and you can see the `#define` for that [here](https://github.com/knolleary/pubsubclient/blob/master/src/PubSubClient.h#L31).

Setting this value below 15sec is probably a bad idea, but that's where you'd do it.

kbweathe, 2020-03-03:
Allen,
I found the file but will take your advice and leave it at 15 sec.  Thanks so much for your help and quick response!  I have wanted to do this for 20+ years.  I also have a sliding gate that I am controlling with your project. I would like to send you some money for a couple of coffees, Cokes, or what ever you do. Can I do that for you? PayPal?  Let me know how and where, if you would accept it. I thank the good Lord for smart people like you so willing to share and help!
Brent

aderusha, 2020-03-04:
> Allen, I found the file but will take your advice and leave it at 15 sec.  Thanks so much for your help and quick response!  

You're welcome!

> I have wanted to do this for 20+ years.  I also have a sliding gate that I am controlling with your project. I would like to send you some money for a couple of coffees, Cokes, or what ever you do. Can I do that for you? PayPal?  Let me know how and where, if you would accept it.

I've added my [Buy Me a Coffee link](https://www.buymeacoffee.com/gW5rPpsKR) to the project page, thanks in advance for the consideration, I really appreciate it!

broyuken, 2020-08-18:
Curious, what is the reason against changing this setting? Would it just be if the car loses wifi while backing in the door could start closing on the roof? I was thinking of changing it to 5 seconds, wondering if this would be a fairly safe compromise?

aderusha, 2020-08-18:
That, and also because WiFi can be spotty, particularly when there is a $1 device with a PCB antenna involved.  So if for whatever reason it goes offline for a few seconds, just the time for the reconnect alone will trip over your 5 second window.  Leave some room for error, particularly when dealing with large mechanical systems.

broyuken, 2020-08-18:
I didn't even think of the time to boot, that makes a lot of sense. I will keep it at 15 seconds, thanks.