aderusha/MQTTCarPresence #23: Can the Arduino code be changed over the air?

VictorMFCoelho, 2021-11-15:
Hi!
My D1 Mini is on a hard to reach spot, can it be accessed over the air?
I have to make changes on two devices, one I have to clear the credencials as the car will be sold and the other a have to change the node name. Can it be done over the air?
Thanks!

aderusha, 2021-11-15:
Kinda yes but probably no.  If you compiled the code yourself and set the [otaPassword](https://github.com/aderusha/MQTTCarPresence/blob/master/MQTTCarPresence/MQTTCarPresence.ino#L5), then yes you can use the arduino Over-The-Air flash update mechanism to flash the device.  If you didn't set this value at compile time, you'll need physical access to the device.

VictorMFCoelho, 2021-11-15:
Thanks a lot!

closed 2021-11-15 by aderusha