aderusha/MQTTCarPresence #26: MQTT autodiscovery

Bluhme1, 2022-05-27:
Hi Thanks for your great work

The car presence module isn't shown in the Mosquito integration.
I get data into MQTT. I can see data in MQTT Explorer, so everything is working ok.

But why can't I see this module in the auto discovered list on Mosquito

/Lars

audioeptesicus, 2022-12-27:
I'm in the same position now.  I am not able to get HA to see the sensors, as auto discover does not seem to find them.

broyuken, 2023-04-03:
if its not finding it, just be aware that it doesn't show up as a device, just entities.

requiem240sx, 2023-06-01:
So Mosquitto has the data, home assistant runs on a different IP.  It doesn't add entities for me in home assistant... I was able to manually add some entities with these configs in my configuration.yaml file... but the blueprint doesn't work with them so I'm assuming I added them incorrectly.  I suppose I could make my own automation at this point, but the behavior seems off, as it only goes on/off as I connect to the MQTT server, I feel like it should be getting other data here. 

```
mqtt:
  binary_sensor:
    - name: "Bike MQTT Connectivity"
      state_topic: "motorcycle/binary_sensor/BikePresence/state"
  sensor:
    - name: "Bike Wifi Signal Strength"
      state_topic: "motorcycle/sensor/BikePresence-signal/state"
      unit_of_measurement: "dBm"
      value_template: "{{ value }}"
    - name: "Bike Device Uptime"
      state_topic: "motorcycle/sensor/BikePresence-uptime/state"
      unit_of_measurement: "ms"
      value_template: "{{ value }}"
```