aderusha/MQTTCarPresence #29: Tasmota Setup

broyuken, 2023-04-03:
I'm trying to get this set up, and it sorta works, but I think it could be done better but I haven't been able to wrap my head around how LWT and Retain works with Tasmota. I use the signal sensor to determine if my car is in the garage, driveway or away and this doesn't work if its unavailable. Do you know of a way to make the tasmota sensor not go unavailable when the car is off and the esp is powered off? Ideally it would show the switch as off as the LWT and the sensor data would be retained. Also I would like to be able to set the keepalive to something shorter as I like to have the delay set in my automation so the door shuts if the switch is off for 15 seconds instead of having the timeout set on the esp8266 itself. 

The main reason I'm trying to use tasmota is because 
1. the arduino method doesn't create a device in HA and
2. This is the only arduino code I use, and would like to consolidate everything in Tasmota or ESPHome for my dev boards.

broyuken, 2023-04-10:
@aderusha Hello, do you know if this is possible?

aderusha, 2023-04-11:
Sorry man, the Tasmota stuff came from a user contribution and I don't use that part of the codebase.  I'd suggest hitting up the Tastmota folks to see if there's a way to make their firmware work in that way.  Wish I had a better answer my friend.

broyuken, 2023-04-11:
All good, this component is awesome as it is. Are there any plans to update it to actually create a device in home assistant as opposed to just entities? If so I’d probably just continue using the arduino code.