aderusha/MQTTCarPresence #30: nodemcu 32s

asucrews, 2023-05-07:
Hello, do you know if anyone has converted this to esp32 based or if it is even possible?

guy0nabuffalo, 2024-03-07:
Hi I recently used chatgpt to convert the code to ESP32. You can find it here https://github.com/guy0nabuffalo/ESP32-MQTTCarPresence

asucrews, 2024-03-09:
Cool, I check this out this weekend

guy0nabuffalo, 2024-03-13:
Let me know how it goes.
Today I put the code on a [ESP32-WROOM-32U](https://www.amazon.com/HiLetgo-ESP32-DevKitC-ESP32-WROOM-32U-ESP-WROOM-32U-Development/dp/B09KLS2YB3/ref=sr_1_3?crid=2Z0VMUPQACU6U&dib=eyJ2IjoiMSJ9.VLMXCdcqQpWNcgWrFEX51fWo1unCG4K2HwzUQftSBWXRFRRg7qxNwZB6q0n7yIVosv5BHqpWPqy80rMNDoqBm8P1_EgOzqetCshLAPjDwXoh67q-VC1qz1Gw3eqWKhKYG3kALdbTIUWJyrlbfTJd6vREk_hdwrNF_pBhDayO6vnzrb0BDouf-z_WZ5L9MLpboGqox1Q5wgByOcvdZv9AnL5B_4lmZU0D6WpuHDU8ZnCbCeZ6D39_vpAo7Y_5Q7NvQ6ewlokVVDXxXTzm5xXOUKsfOmDLF4SBYsm8DqnpWGw.nkiOcYWfKhjago7RcCTcbvyld_b0KJh2FnN5pPvtOak&dib_tag=se&keywords=esp32-devkitc+wroom-32u&qid=1710354144&s=electronics&sprefix=esp32-devkitc+wroom-32u%2Celectronics%2C109&sr=1-3) which uses an [external antenna](https://www.amazon.com/Antenna-Pigtail-Wireless-Routers-Repeater/dp/B07R21LN5P/ref=sr_1_1_sspa?crid=1NJY40YESVVWE&dib=eyJ2IjoiMSJ9.WHDYYAkQkOMd38Qa3sPwXwwJyo5f_4w9-XXYM68Axe3PM1JlTiMP1MNvkr9qzcqH7UD0mLB8fUxPE0N-UMSkD2qP7jS0sxeRv6LjS0_qeB_J9yBVlPLouojSTNaGpPwIQwJmMVwn1FSHKBjcgx7iTnsmd3i8XkIfWrLtLLP1-WouS0IOrgdq-oKHalHKWKyStMEl-h2K3_pByp5RnVB9iydgo75sPbj1PeKrXiFfw7lr6DHmLo6K2sRILOFIFqDrvalDAD7a30kAXU723labpPnqq9c3wSEVEkax7cOLT2o.hsjTPpEp0EnTBls7wUbMv32HLD_o2a16ytxe--wIKSA&dib_tag=se&keywords=Wueenti+wifi+RP-SMA+male+antenna&qid=1710354294&s=electronics&sprefix=wueenti+wifi+rp-sma+male+antenna%2Celectronics%2C102&sr=1-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1). Seems to be working well. Range is better than an ESP32 with the onboard antenna.