aderusha/MQTTCarPresence #31: Confirmation of expected behavior: do nothing if state is already matching desired state

xanderphillips, 2023-12-27:
My garage doors will close on their own using the timer in the 'smart' opener from the manuf.
I have ratgdo setup to provide state and control to HA.  Can someone confirm if the blueprint automation is smart enough to not try to close an already closed door?  I am guessing by the way the logic looks to my untrained eyes that it should ignore trying to close an already closed door, but I'd like a second opinion.

I guess same would go for not trying to open an already open door.

aderusha, 2023-12-27:
This probably doesn't require two issues :D

The "delay" is entirely handled by your cover device.  For example, the blueprint here calls "cover.open" if the device has come online and the door is currently "closed".  A cover can be "open" or "opening" or "closed" or "closing" (or a percentage if supported).  How your cover progresses through those stages is up to whatever integration is handling your door and is outside of the scope or control of the blueprint here.

xanderphillips, 2023-12-27:
That's the confirmation I was looking for, that it looks at the current state, and not just blindly sending an open if it's already open or a close if it's already closed, as I believe my door would toggle instead of honoring the actual command. I need to test that actually!  Thanks for the quick response!

closed 2023-12-27 by xanderphillips