aderusha/MQTTCarPresence #33: How can I add additional conditions to the initial car start up 'open garage'?

NZ3-14, 2024-03-27:
Hi,

Love the solution - Thanks for making it available!

I'm new to HA and moving from Homeseer.

How can I add two additional conditions to the initial car starting-up sequence?

1—Test that a light is on in the garage (I have a motion-activated light in the garage). This will confirm that I am in the garage, as the car wakes up at times to report its state of charge, and this wake-up activates the USB port.

2- Test that the car is in the garage (open-garage distance sensor) as the car is often left on the driveway and is driven away.

Huge thanks for any assistance!

Philip

aderusha, 2024-03-27:
You'd have to write your own automations for that, but you can use the code presented in the blueprint to get you started.  Should all be pretty reasonable to make happen, the blueprint itself doesn't do all that much and should be easy to re-create and then add to using the HA automation editor.

NZ3-14, 2024-03-30:
Thanks - Can you please let me know at which line I should add the conditions to limit the opening hours?